OpenAgricultureFoundation/openag-device-software #165: Investigate ubuntu core / snaps

jakerye, 2018-10-24:
Ubuntu core looks like a great way to never brick devices. The entire OS is "containerized." 

https://www.ubuntu.com/core

Also comes with a great package management system that runs containerized apps which can easily be versioned up / down.

https://snapcraft.io/

This isn't the highest priority item as we already have a debian-based packaging system implemented...just keeping in mind for future development...

rbaynes, 2018-10-24:
> Gotcha...i see it is not listed in their promoted hardware:
> https://developer.ubuntu.com/core/get-started
>
> In that case, it probably does not make much sense for a bbb-based system

yeah, I think for an ARM chip, we are stuck with the debian image sponsored
/ tested by beaglebone.

Thanks,
Rob Baynes
*pronouns: he / him / his*

jakerye, 2018-10-24:
Not sure I follow for an ARM chip? RasPi and Dragonboard 410C both run on ARM...I do agree though that the best OS choice for beaglebone is the debian image sponsored / tested by beaglebone

jakerye, 2018-10-24:
Really just posted this cause I think it is a cool system and could be a good consideration for future development efforts...not trying to make any large architectural / platform changes right now...

jakerye, 2018-10-31:
Closing since not on the near term horizon for bbb development...still a good technology to keep in mind...

closed 2018-10-31 by jakerye