OpenAgricultureFoundation/openag-device-software #169: integrate image processing algorithms to detect low hanging fruit

jakerye, 2018-10-31:
This is challenging to get reliable but it would be great to implement some of the "demo" algorithms into the code base. 

There has already been work that detects number of leaves, plant width, etc.

There needs to be some pre-processing filters to determine if an image is valid to be processed. The main concern is to detect occlusion (what happens when the plants get big and they overlap?) Can we put the plant site locations in the device config file? Then make sure the image has isolated clusters around each plant site. Then if there is not occlusion, run through the leaf count processor. The leaf count processor will likely need to know the cultivar classification (Basil leaves vs. Arugula leaves).

jakerye, 2018-10-31:
See work done previously by Eddie

wsokno, 2018-10-31:
Maybe you could you tag the plant sites with some kind of unique fiducial marker like a [2D barcode](https://en.wikipedia.org/wiki/ShotCode). That might also help with identifying cameras for issue #168 (multiple cameras).