OpenAgricultureFoundation/openag-device-software #174: add heartbeat for sending light commands to make more robust against power blips

jakerye, 2018-11-07:


rbaynes, 2018-11-08:
I have been running the master branch (type-check) on my EDU and I notice
that several times a day, the LEDs go off for 30 seconds or so, then come
back on.  Is this a power blip?

Thanks,
Rob Baynes
*pronouns: he / him / his*

jakerye, 2018-11-08:
hmm...not sure what exactly it is...I know that the bbb is currently using the hack to periodically resets itself to re-initialize the usb camera so perhaps it is that? if not it very well could be the power blip...if the capacitors are not populated near the dac on your edu brain then it is possible it is the power blip...

jakerye, 2018-11-08:
fixed in 474710eebb669d28086f42326f9a1833680ce657

closed 2018-11-08 by jakerye