OpenAgricultureFoundation/openag-device-software #176: recheck for verification code in iot manager

jakerye, 2018-11-08:
after initial registration iot manager does not pick up the verification code until a reboot occurs...perhaps the script runs in the background so the function returns before complete so when it initially looks for a registration code it still gets unknown...

can also spin while waiting for device verification code to be not "UNKNOWN" and throw an exception upon a timeout

rbaynes, 2018-11-08:
That was working well.   Perhaps with the code re-org we no longer wait for
the script to run that produces the verification.txt file we need to read?

Thanks,
Rob Baynes
*pronouns: he / him / his*

jakerye, 2018-11-08:
Yep I expect it will just be a quick & easy fix with the method mentioned above. I'll roll it in shortly.

jakerye, 2018-11-08:
fixed in 839d22c5410ae0a5a56e52d3ed3be16856eee786

closed 2018-11-08 by jakerye