OpenAgricultureFoundation/openag-device-software #178: download csv from device ui

jakerye, 2018-11-14:


utkarshdeep, 2019-03-22:
Hey @jakerye. I am very excited about the OpenAg initiative and reading a lot on the forum as well. I am willing to contribute. If you can elaborate on the issue here, I can take it forward and resolve this.

rbaynes, 2019-03-22:
Hi @utkarshdeep.   Jake has left the project, but there are two other engineers still perfecting it.

The intent of this issue was to allow a user to download a CSV file of data the device is producing.
The things that would have to happen are:

1.  Where the data is being published by the IoTManager, also write it into the database (which has recently switched from postgres to sqlite).
2. Make sure we are not filling up the DB without bounds, the data should be written to only keep the last 100 or so values of each variable.
3. Add a button to the UI to create and download the CSV data.

Thank you for the interest and help!

utkarshdeep, 2019-03-22:
I am sure the two other engineers are doing a great job.

If there is any other issue I can help with, please let me know, I am more than willing to help!