OpenAgricultureFoundation/openag-device-software #182: Default 0x47 address for DAC5578 is not valid according to datasheet

wsokno, 2018-11-19:
I wonder if I'm overlooking something obvious, but I've checked this over several times now. I can't figure out how to resolve the I2C address discrepancy for the DAC5578. I wonder if maybe your LED dimming code has a mix of stuff from different hardware revisions.

In [device/peripherals/modules/led_dac5578/setups](https://github.com/OpenAgInitiative/openag-device-software/tree/master/device/peripherals/modules/led_dac5578/setups), the default DAC5578 I2C slave addresses [are](https://github.com/OpenAgInitiative/openag-device-software/blob/8a0335293be00d9d8fc6f3e8af1fff790f1cffa8/device/peripherals/modules/led_dac5578/setups/libra-v1.json#L56) [all](https://github.com/OpenAgInitiative/openag-device-software/blob/8a0335293be00d9d8fc6f3e8af1fff790f1cffa8/device/peripherals/modules/led_dac5578/setups/orion-v1.json#L55) [defined](https://github.com/OpenAgInitiative/openag-device-software/blob/8a0335293be00d9d8fc6f3e8af1fff790f1cffa8/device/peripherals/modules/led_dac5578/setups/taurus-v1.json#L55) [as](https://github.com/OpenAgInitiative/openag-device-software/blob/8a0335293be00d9d8fc6f3e8af1fff790f1cffa8/device/peripherals/modules/led_dac5578/setups/taurus-v2.json#L55) `0x47`. But, on page 31 of the [DAC5578 datasheet](http://www.ti.com/lit/ds/symlink/dac5578.pdf), it says that the possible I2C slave addresses for the TSSOP packages are `1001 000` (`0x48`), `1001 010` (`0x4a`), or `1001 100` (`0x4c`), depending on whether the ADDR0 pin is low, high, or floating.

`0x47` is not a valid slave address for the DAC5578. On the other hand, there's a test script at [device/peripherals/common/dac5578/tests/test_driver.py](https://github.com/OpenAgInitiative/openag-device-software/blob/master/device/peripherals/common/dac5578/tests/test_driver.py) that's expecting to talk to the DAC at [address `0x4c`](https://github.com/OpenAgInitiative/openag-device-software/blob/8a0335293be00d9d8fc6f3e8af1fff790f1cffa8/device/peripherals/common/dac5578/tests/test_driver.py#L26). There's another script at [scripts/test_edu002.sh](https://github.com/OpenAgInitiative/openag-device-software/blob/8a0335293be00d9d8fc6f3e8af1fff790f1cffa8/scripts/test_edu002.sh) that mentions "EDU 002" and uses `i2cset -y 2 0x47 ...` extensively. Is that perhaps left over from when you were evaluating the PCA9685 PWM driver? According to page 8 of the [PCA9685 datasheet](https://www.nxp.com/docs/en/data-sheet/PCA9685.pdf), it looks like it would be possible to set it up for address `0x47` using its address pins, although that isn't the default address.

Do you know what's going on?

### Context Screenshots

**From p31 of the DAC5578 datasheet:**

[image]

**Binary to hex conversions:**
[image]

**From taurus-v2.json:**
[image]

**From test_driver.py:**
[image]

**From test_edu002.sh:**
[image]

jakerye, 2018-11-19:
I completely understand your confusion. 0x47 is the broadcast address in case there are multiple dacs on the same I2C bus we want to talk to at the same time. It is useful for multiple panels where independent control would needlessly slow down the comms. Main caveat with using broadcast is the inability to read from a device but haven't found a need for doing that yet so all devices that use a dac default to using the broadcast address.

In practice, 2 addresses show up for the dac upon an i2c scan. The individual address (one of the 3 you mentioned above) and the broadcast address.

Here is a screenshot of an i2c scan on a beaglebone (don't worry about 0x77 -- that is the i2c mux address)

[image]

wsokno, 2018-11-19:
Okay, thanks. Indeed I was missing something obvious. I see that now in the table on page 33.

If you used individual addresses, you could periodically read out the channel settings to detect the power glitches you've referred to here in other issues.

closed 2018-11-19 by wsokno