OpenAgricultureFoundation/openag-device-software #251: Create development image for pfc4

jakerye, 2019-07-30:


YusufKhan, 2019-09-09:
Can the image be added to the release of its branch in this repo?

rbaynes, 2019-09-09:
Hi Yusuf,
We are using docker on the balena cloud to deploy images now.  So all we
have is a docker set up file and the image is automatically updated.

This repo can run stand alone on any Linux or OSX machine.   I recommend
using it that way for now.

srmoore, 2019-09-09:
Once the PFC4 is stable, we'll look into releasing an image that runs without Balena for developers / people not managing larger deployments.  As @rbaynes said, the current state can be just pulled and run (once the dependencies are installed) on Linux or OSX now though.

YusufKhan, 2019-09-09:
Thanks for clarifying. I'll install it on Linux, and I assume this is the pfc4-debug branch?
Also, does this mean that the v2.0.1 image (Google Drive link from installation instructions) is no longer suitable for testing?

rbaynes, 2019-09-09:
Please use the master branch.  It is our "stable and ready to deploy"
version.  Any other branches are for features or bug fixing or works in
progress, so not to be used without incurring some risk.

Yes, that image and install instruction doc are out of date.  We will fix
the docs once we get this big push on the v4 done, sorry for the confusion.

Thanks,
Rob Baynes
*pronouns: he / him / his*